NicoHood/PinChangeInterrupt #1: 'PCICR' not declared for ATtiny44

chbmuc, 2015-05-13:
I was trying to use this library with an ATtiny44, however I get:

> PinChangeInterrupt.cpp: In function 'void attachPinChangeInterruptHelper(uint8_t, uint8_t, uint8_t)':
> PinChangeInterrupt.cpp:117:2: error: 'PCICR' was not declared in this scope
>   PCICR |= (1 << pcintPort);
>   ^
> PinChangeInterrupt.cpp: In function 'void detachPinChangeInterruptHelper(uint8_t, uint8_t)':
> PinChangeInterrupt.cpp:134:3: error: 'PCICR' was not declared in this scope
>    PCICR &= ~(1 << pcintPort);

I think this is because the ATtiny has no PCICR.

NicoHood, 2015-05-13:
You are right. Its probably
9.3.2         GIMSK – General Interrupt Mask Register

http://www.atmel.com/Images/doc8006.pdf
I will work on a fix...

NicoHood, 2015-05-16:
Can you try the current dev branch?
https://github.com/NicoHood/PinChangeInterrupt/tree/dev

chbmuc, 2015-05-17:
I can confirm, that it compiles, but I realized that the ATtiny X5 need some additional #ifdefs:

```diff
--- PinChangeInterrupt.cpp.orig 2015-05-17 21:10:53.077443193 +0200
+++ PinChangeInterrupt.cpp      2015-05-17 19:46:11.446594600 +0200
@@ -110,14 +110,27 @@
        // update the old state to the actual state
        oldPorts[arrayPos] = *portInputRegister(pcintPort);

-       // pin change mask registers decide which pins are ENABLE as triggers
+// pin change mask registers decide which pins are ENABLE as triggers
+#ifdef PCMSK0
        *(&PCMSK0 + pcintPort) |= pcintMask;
-
+#elif defined(PCMSK) /* e.g. ATtiny X5 */
+  *(&PCMSK + pcintPort) |= pcintMask;
+#else
+#error MCU has no PCMSK register
+#endif
+
        // PCICR: Pin Change Interrupt Control Register - enables interrupt vect                        ors
 #ifdef PCICR
        PCICR |= (1  << (pcintPort + PCIE0));
-#elif defined(GIMSK)
+#elif defined(GIMSK) /* e.g. ATtiny XX */
+#ifdef PCIE0
        GIMSK |= (1  << (pcintPort + PCIE0));
+#elif defined(PCIE) /* e.g. ATtiny X5 */
+  GIMSK |= (1  << (pcintPort + PCIE));
+  sei();
+#else
+#error MCU has no PCIE register
+#endif
 #else
 #error MCU has no such a register
 #endif
@@ -133,14 +146,30 @@
        fallingPorts[arrayPos] &= ~pcintMask;

        // disable the mask.
+#ifdef PCMSK0
        *(&PCMSK0 + pcintPort) &= ~pcintMask;
-
+
        // if that's the last one, disable the interrupt.
        if (*(&PCMSK0 + pcintPort) == 0)
+#elif defined(PCMSK) /* e.g. ATtiny X5 */
+  *(&PCMSK + pcintPort) &= ~pcintMask;
+
+  // if that's the last one, disable the interrupt.
+       if (*(&PCMSK + pcintPort) == 0)
+#else
+#error MCU has no PCMSK register
+#endif
+
 #ifdef PCICR
                PCICR &= ~(1  << (pcintPort + PCIE0));
-#elif defined(GIMSK)
+#elif defined(GIMSK) /* e.g. ATtiny XX */
+#ifdef PCIE0
                GIMSK &= ~(1  << (pcintPort + PCIE0));
+#elif defined(PCIE) /* e.g. ATtiny X5 */
+               GIMSK &= ~(1  << (pcintPort + PCIE));
+#else
+#error MCU has no PCIE register
+#endif
 #else
 #error MCU has no such a register
 #endif
```

I'm having some trouble with my circuit, so I'm not sure if it actually works, but I think this has nothing to do with this library.

chbmuc, 2015-05-17:
Found the problem with my circuit: Everything works fine. (Tested on an ATtiny85 with the patch from above).

NicoHood, 2015-05-18:
Can you create a PR with the fix that works for you? Please compare with the dev branch. That would be nice. Thanks.

NicoHood, 2015-05-20:
I will try to publish a new version release to the master soon. I have some problems with my local git, so use the dev branch for now. Thanks chbmuc!

NicoHood, 2015-05-24:
Now online in the master branch.
Can you please check if it works?

chbmuc, 2015-07-02:
Sorry for the late response: It works!

closed 2015-07-02 by chbmuc