NicoHood/PinChangeInterrupt #4: ATTiny13A support

cb88, 2015-08-27:
It would be really nice to be able to use this library... I'm using the core13 toolchain as it produces the smallest .hex at least for the program I wrote for it anyway. 

http://sourceforge.net/projects/ard-core13/

NicoHood, 2015-08-27:
Can you try this branch?
https://github.com/NicoHood/PinChangeInterrupt/tree/NicoHood-Attiny13

cb88, 2015-08-27:
```
C:\<snip>\Arduino-1.6.5\libraries\PinChangeInterrupt\src\PinChangeInterrupt.cpp: In function 'void attachPinChangeInterruptHelper(uint8_t, uint8_t, uint8_t)':
C:\<snip>\Arduino-1.6.5\libraries\PinChangeInterrupt\src\PinChangeInterrupt.cpp:111:51: error: 'portInputRegister' was not declared in this scope
  oldPorts[arrayPos] = *portInputRegister(pcintPort);
                                                   ^
Error compiling.
```

NicoHood, 2015-08-28:
This seems to be missing inside your core:
https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/cores/arduino/Arduino.h#L174

Try to uncomment this line and recompile. It should do the job anyways. Its not a good fix, but I want to see if it works in general. This is a minimal improvement.

I will try to find a way to work around this.

NicoHood, 2015-08-28:
Try again please:
https://github.com/NicoHood/PinChangeInterrupt/tree/NicoHood-Attiny13

cb88, 2015-08-28:
I was able to build it against the core13 toolchain. I'll test it out monday to make sure it works as intended!

NicoHood, 2015-10-03:
Any tests made yet?

cb88, 2015-10-05:
I'll have to get around to it. I currently am just reading the IO in a loop... I think it still didn't work last I checked. But I'll have to get it back out and test it again... unfortunately it's part of a project at work that got made low priority so it might be a few days until I have time.

NicoHood, 2015-10-05:
Well you can test the led example and see if it works. This avoids any errors on your code :)

NicoHood, 2015-12-05:
I am closing this now, since you never gave me any feedback. Attiny13 should work now.

closed 2015-12-05 by NicoHood