NicoHood/PinChangeInterrupt #8: Problem getting Pin Change Interrupts to work on ATtiny84

PaulRB, 2016-03-20:
I have been trying to get Pin Change interrupts to work on an ATtiny84.

Code: 

```
#include "PinChangeInterrupt.h"

#define BUTTON  2
#define LED 8

void setup() {

  pinMode(BUTTON, INPUT_PULLUP);
  pinMode(LED, OUTPUT);
  attachPinChangeInterrupt(digitalPinToPinChangeInterrupt(BUTTON), buttonInterrupt, CHANGE);
 
}

void buttonInterrupt() {
  digitalWrite(LED, digitalRead(BUTTON));
}

void loop() {

}

```
With the code shown above, nothing happens when the button is pressed or released. However, if I move the digitalWrite() line of code out of the sensorInterrupt() function and into loop(), the led goes off and on perfectly as the button is pressed and released. So it seems the interrupt function is not getting called.

I have tried the same code on Nano 3 and that works perfectly, the interrupt function gets called.

I am using:

-  IDE 1.6.5
-     Tiny core: https://raw.githubusercontent.com/damellis/attiny/ide-1.6.x-boards-manager/package_damellis_attiny_index.json. Have tried version 1.0.0 and 1.0.1
-     Pin change library: https://github.com/NicoHood/PinChangeInterrupt version 1.2.3

NicoHood, 2016-03-20:
Which `sensorInterrupt()` function? Can you post a working and a not working example so we can compare better?

Also I suggest to upgrade the IDE, it is quite old.

PaulRB, 2016-03-20:
Apologies, I meant to say buttonInterrupt() !
1.6.5 is only one or two versions behind, I will give 1.6.7 a go and let you know.
Worgink version: move the digitalWrite() into loop() and that works. Alternatively, it works as shown on Nano 3.

NicoHood, 2016-03-20:
Then it is a bug of the library.
Can you please double check all wires? Maybe the pin numbers are wrong?

Can you please test:
* Pull every button low with the current setup and see if the led changes (wrong cables used)
* Test all pins, aka change the code (partly wrong library definitions)

Otherwise I have to double check the 84 definitions

PaulRB, 2016-03-20:
Thanks @NicoHood 

Like I am saying, if I move the digitalWrite() into loop(), it works, with no changes to wiring, so the wiring must be OK.

I will try using other pins for the button.

NicoHood, 2016-03-20:
Oh. Right. Good Point.
Testing other buttons would be nice, as well as getting the output of the definition.

PaulRB, 2016-03-20:
Just tried pin 0 on the tiny84. Same issue. once again, works fine if I move digitalWrite() into loop()

NicoHood, 2016-03-20:
Does `Serial.print(PCINT0);` give an error?

PaulRB, 2016-03-20:
Well, this is a tiny84, so Serial.print() won't be available, will it?

NicoHood, 2016-03-20:
`int i = PCINT0;`

PaulRB, 2016-03-20:
Only this warning:
`/home/paul/sketchbook/PCInterruptTest/PCInterruptTest.ino: In function 'void loop()':
/home/paul/sketchbook/PCInterruptTest/PCInterruptTest.ino:19:7: warning: unused variable 'i' [-Wunused-variable]
   int i = PCINT0;
       ^`

NicoHood, 2016-03-20:
I have no idea why it then should not work. I'd have to further look into this issue. Please test the other buttons ans report the result. If you have an 85 please also test this mcu.

PaulRB, 2016-03-20:
OK, will do. Maybe later this eve or tomorrow. I do have tiny85 also. I will also try to find the value of digitalPinToPCINT(BUTTON) by some method. I have the tiny84 wired as an i2c slave to another Arduino, so will try passing the value to the master to print to serial.

PaulRB, 2016-03-21:
I tried my sketch above with all pins 0 - 10 on the tiny84. None of them worked, sorry.

But I have managed to get your debug script working, using [https://github.com/jscrane/TinyDebugSerial](TinyDebugSerial).

Here is the sketch I uploaded:
`
/*
 Copyright (c) 2014 NicoHood
 See the readme for credit to other people.

 PinChangeInterrupt_Development
 Prints the available PCINTs.
 This is not the Pin number!
 Only useful for developing and checking board definitions.
 */

#include "PinChangeInterrupt.h"

#include "TinyDebugSerial.h"

TinyDebugSerial mySerial = TinyDebugSerial();

void setup()
{
  // start serial debug output
  mySerial.begin(9600);

}

void loop() {
  mySerial.println("Available/used pins");
  mySerial.println(PCINT_USE_PCINT0);
  mySerial.println(PCINT_USE_PCINT1);
  mySerial.println(PCINT_USE_PCINT2);
  mySerial.println(PCINT_USE_PCINT3);
  mySerial.println(PCINT_USE_PCINT4);
  mySerial.println(PCINT_USE_PCINT5);
  mySerial.println(PCINT_USE_PCINT6);
  mySerial.println(PCINT_USE_PCINT7);
  mySerial.println(PCINT_USE_PCINT8);
  mySerial.println(PCINT_USE_PCINT9);
  mySerial.println(PCINT_USE_PCINT10);
  mySerial.println(PCINT_USE_PCINT11);
  mySerial.println(PCINT_USE_PCINT12);
  mySerial.println(PCINT_USE_PCINT13);
  mySerial.println(PCINT_USE_PCINT14);
  mySerial.println(PCINT_USE_PCINT15);
  mySerial.println(PCINT_USE_PCINT16);
  mySerial.println(PCINT_USE_PCINT17);
  mySerial.println(PCINT_USE_PCINT18);
  mySerial.println(PCINT_USE_PCINT19);
  mySerial.println(PCINT_USE_PCINT20);
  mySerial.println(PCINT_USE_PCINT21);
  mySerial.println(PCINT_USE_PCINT22);
  mySerial.println(PCINT_USE_PCINT23);
  mySerial.println();

  mySerial.println("Available/used pins per port");
  mySerial.println(PCINT_NUM_PINS_PORT0);
  mySerial.println(PCINT_NUM_USED_PINS_PORT0);
  mySerial.println(PCINT_NUM_PINS_PORT1);
  mySerial.println(PCINT_NUM_USED_PINS_PORT1);
  mySerial.println(PCINT_NUM_PINS_PORT2);
  mySerial.println(PCINT_NUM_USED_PINS_PORT2);
  mySerial.println();

  mySerial.println("Available/used ports");
  mySerial.println(PCINT_HAS_PORT0);
  mySerial.println(PCINT_USE_PORT0);
  mySerial.println(PCINT_HAS_PORT1);
  mySerial.println(PCINT_USE_PORT1);
  mySerial.println(PCINT_HAS_PORT2);
  mySerial.println(PCINT_USE_PORT2);
  mySerial.println();
  delay(5000);
}`

And here is the output:
`Available/used pins
1
0
1
1
1
1
1
1
1
1
1
1
0
0
0
0
0
0
0
0
0
0
0
0

Available/used pins per port
8
7
4
4
0
0

Available/used ports
1
1
1
1
0
0

`

NicoHood, 2016-03-21:
Great. thx for the output. I will have a look later this day. Remind me if i forget. You can also tox me: www.contact.nicohood.de

PaulRB, 2016-03-21:
In case you need it, here is the debug output from the tiny85:
`Available/used pins
1
1
1
1
1
1
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0

Available/used pins per port
6
6
0
0
0
0

Available/used ports
1
1
0
0
0
0

`

PaulRB, 2016-03-21:
Well, don't know if this is good news or bad news, but I can't get PC Interrupts to work on any pin of tiny85 either.
[image]

PaulRB, 2016-03-21:
Just to reassure you that the library is not completely broken, I just tested it with Nano 3, using pins 2, 3, 13, A0... works perfectly each time.
[image]

NicoHood, 2016-03-21:
Thx. The error occurs on 85 is better to debug.
I now tried to use my trinket (85) to debug this, but the code compiles for the 328 for some reason. Maybe this is also your problem? You have to enable the compiler output and check the -mmcu option. But i cannot even upload the program.

So i have to ask you for now until i get this fixed:
* connect leds to pin 0, 1, 2
* upload the sketch and trigger the button, report what leds are blinking
* uncommend the comment and do not touch anything. report what the leds are doing
```cpp
#include "PinChangeInterrupt.h"

// choose a valid PinChangeInterrupt pin of your Arduino board
#define pinBlink 3

void setup() {
  // set pin to input with a pullup, led to output
  pinMode(pinBlink, INPUT_PULLUP);
  pinMode(2, OUTPUT);
  pinMode(0, OUTPUT);
  pinMode(1, OUTPUT);

  // attach the new PinChangeInterrupt and enable event function below
  attachPCINT(digitalPinToPCINT(pinBlink), blinkLed, CHANGE);
}

ISR(PCINT0_vect);
void blinkLed(void) {
  // switch Led state
  digitalWrite(2, !digitalRead(2));
}

void loop() {
  // nothing to do here
  delay(1000);
  digitalWrite(0, !digitalRead(0));

  // uncomment for 2nd test
  //PCINT0_vect();
}
```

PaulRB, 2016-03-21:
Hi.

The sketch blinks the led on pin 0, 1s on, 1s off. Pressing the button has no effect on any led.

Then I uncomment the final code line for second test. There is no change in behavoir of the tiny84.

Finally I add the line `digitalWrite(1, digitalRead(pinBlink));` into loop(). Now, the led on pin 1 lights. If I press and hold the button, pin 1 led goes off (after 1s delay). When I release the button, pin 1 led lights again (after 1s).
[image]

NicoHood, 2016-04-16:
Thanks for all the help! You can retest, but not it all should work fine (I deactivated those 4 pins by default again, to safe flash) :)

closed 2016-04-16 by NicoHood