NicoHood/PinChangeInterrupt #12: Activate pin on Attiny84

tguillaume02, 2016-09-14:
Hello,
I would like use pin PB2 on Attiny84 but I don't know how make.
You say :
>  You may activate them in the setting file (advanced).

But where is this file and how make ?
Thank you

NicoHood, 2016-09-15:
https://github.com/NicoHood/PinChangeInterrupt/blob/master/src/PinChangeInterruptSettings.h#L221

tguillaume02, 2016-09-15:
ok, but pin INT0 is not disable by default?
because I try 
`attachPinChangeInterrupt(digitalPinToPCINT(2),check_btn,FALLING);  `

but it doens't work
```   	
                      _________
      	              |  |_|  |
                      |       |
                VCC 1 |       | 14  GND
      (D 0) PB0     2 |  	  | 13  AREF ( D10)
      (D 1) PB1     3 |	      | 12  PA 1 (D 9)
                    4 |       | 11  PA2 (D 8)
    INT0 (D 2) PB2  5 |	      | 10 PA 3 (D 7)
      (D 3) PA7     6 |    	  | 9   PA 4 (D 6)
      (D 4) PA6     7 |    	  | 8  PA 5 ( D5)
     		          |_______|
```

NicoHood, 2016-09-15:
It should not. please try all pins and report which work an which not. I've had bugs in the code before for attiny, so its possibly that i've missed this. I've never tested the attnys myself.

tguillaume02, 2016-09-15:
I have check all pin with this code 

````
#include <PinChangeInterrupt.h> // http://code.google.com/p/arduino-tiny/downloads/list

void blinks()
{

  digitalWrite(3, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);              // wait for a second
  digitalWrite(3, LOW);    // turn the LED off by making the voltage LOW
  delay(1000);              // wait for a second
}

// the setup function runs once when you press reset or power the board
void setup() {
  // initialize digital pin 13 as an output.
  pinMode(3, OUTPUT);
 attachPinChangeInterrupt(digitalPinToPCINT(2),blinks,FALLING);   // attache l'interruption externe n°0 à la fonction check_btn 
  
}

// the loop function runs over and over again forever
void loop() {
}
````

Pins D0, D1, D2 didn't work, other pins are ok.

NicoHood, 2016-09-16:
Have you enabled PortB a suggested?

tguillaume02, 2016-09-17:
Yes I have change this code 
```
#if defined(__AVR_ATtiny24__) || defined(__AVR_ATtiny44__) || defined(__AVR_ATtiny84__)
// Port1 is connected to reset, crystal and Pin Interrupt 0
// Deactivate it by default
#if defined(PCINT_ENABLE_PORT1)
#undef PCINT_ENABLE_PORT1
#endif
#endif
```

IN 
```
if defined(__AVR_ATtiny24__) || defined(__AVR_ATtiny44__) || defined(__AVR_ATtiny84__)
// Port1 is connected to reset, crystal and Pin Interrupt 0
// Deactivate it by default
if defined(PCINT_ENABLE_PORT1)
undef PCINT_ENABLE_PORT1
endif
endif
```

NicoHood, 2016-09-17:
This is expected. You need to  comment the undef:
```cpp
#if defined(__AVR_ATtiny24__) || defined(__AVR_ATtiny44__) || defined(__AVR_ATtiny84__)
// Port1 is connected to reset, crystal and Pin Interrupt 0
// Deactivate it by default
#if defined(PCINT_ENABLE_PORT1)
//#undef PCINT_ENABLE_PORT1
#endif
#endif
```

closed 2017-05-11 by NicoHood