NicoHood/PinChangeInterrupt #14: Not Compiling with SoftwareSerial

ArduinoFanBoy, 2017-05-11:
Hi there, I am using a PCINT on A1 (PCINT9 according to the table) on an Arduino Uno but this library doesn't compile with the SoftwareSerial library because it also uses interrupts. I get this error:

`libraries\SoftwareSerial\SoftwareSerial.cpp.o: In function `__vector_3':

C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SoftwareSerial\src/SoftwareSerial.cpp:229: multiple definition of `__vector_4'

libraries\PinChangeInterrupt\PinChangeInterrupt.a(PinChangeInterrupt.cpp.o):C:\Users\Timothy Woo\Documents\Arduino\libraries\PinChangeInterrupt\src/PinChangeInterrupt1.cpp:40: first defined here

collect2.exe: error: ld returned 1 exit status`

How can I get rid of this error?

NicoHood, 2017-05-11:
This is intended, both libraries are not compatible with each other. I tried to solve this, but there is no real way to do so. You might want to switch to an MCU with more serials or more normal interrupts.

ArduinoFanBoy, 2017-05-11:
I actually​ found (with the help of someone on the AVR forum) that if you know which pin you are using for the PCINT library, you can comment out line 235 or similar in SofrtwareSerial.cpp and it works. As long as there is no port conflict.

closed 2017-05-11 by ArduinoFanBoy