NicoHood/PinChangeInterrupt #15: Does not work above pin 7 on ATmega1284P

savnik, 2017-06-19:
The library does not work for pin 8 and above on the ATmega1284P.
I have been using the pinchangeinterrupt_led example to test. It works fine below pin 8 and at pin 8 and above the interrupt is not activated.

NicoHood, 2017-06-19:
Did you try every available pin?

savnik, 2017-06-20:
No not litterally every pin. But tested pin D0-7 with good results then pin (D)8, 9, 12, 13, 14, 15 with bad results. 
A0 - A7 works also fine.

savnik, 2017-06-20:
I just discovered that it it was only on port D that did not work. 

In the pinchangeinterrupt_howItWorks i got it working by adding

#elif (PCIE == 3)
#define PCINT_vect PCINT3_vect

Into the PCINT definitions

#if (PCIE == 0)
#define PCINT_vect PCINT0_vect
#elif (PCIE == 1)
#define PCINT_vect PCINT1_vect
#elif (PCIE == 2)
#define PCINT_vect PCINT2_vect
#elif (PCIE == 3)
#define PCINT_vect PCINT3_vect
#else
#error This board doesnt support PCINT ?
#endif

NicoHood, 2017-06-20:
It should support all 32 PCINTs, but I never got any hardware to test:
https://github.com/NicoHood/PinChangeInterrupt/blob/master/src/PinChangeInterrupt.h#L141

Can you prepare a pullup request so i can directly review your changes?

NicoHood, 2017-07-03:
Fixed in https://github.com/NicoHood/PinChangeInterrupt/commit/024264c42d792d5b4a8deb5bdde086462bcb0928
Please test this!

NicoHood, 2017-09-02:
1.2.5 was released. It should be fixed now.

closed 2017-09-02 by NicoHood