NicoHood/PinChangeInterrupt #17: ATTiny85 different interrupt pins only respond to different HIGH/LOW states

Bwanna, 2017-07-03:
This is with a Digispark ATTiny85 having two input buttons, each coded to trigger a different routine.  THIS PART IS WORKING FINE.
ISSUE: one button ONLY works when changed to HIGH, the other button ONLY works when changed to LOW.

Sketch explanation:  One button activates a slow cycle rate, the other button activates a fast cycle rate.

#include "PinChangeInterrupt.h"
#define pinBlink1 3     // PhysPin 2
#define pinBlink2 4     // PhysPin 3
#define relayPin 1      // Phys Pin 6

void setup() {
  pinMode(pinBlink1, INPUT_PULLUP);
  pinMode(pinBlink2, INPUT_PULLUP);
  pinMode(relayPin, OUTPUT);
  attachPCINT(digitalPinToPCINT(pinBlink1), blinkLed1, CHANGE);
  attachPCINT(digitalPinToPCINT(pinBlink2), blinkLed2, CHANGE);
}

void blinkLed1(void) {    // Slow rate
  // Switch Relay state
   for (int i = 0; i <= 5; i++) {
    digitalWrite(relayPin, HIGH);
    delayMicroseconds(60000);
    digitalWrite(relayPin, LOW);
    delayMicroseconds(60000);
    }
}

void blinkLed2(void) {   //Fast rate
  // Switch Relay state
   for (int i = 0; i <= 10; i++) {
    digitalWrite(relayPin, HIGH);
    delayMicroseconds(6000);
    digitalWrite(relayPin, LOW);
    delayMicroseconds(6000);
    }
}

void loop() {
  // Nothing to do here
}

NicoHood, 2017-07-03:
Thanks for reporting. As said in chat, I dont have an answer to this yet, I assume its a problem on your site. I will test once I have hardware in place. If anyone else has a similar issue, please let me know.

Bwanna, 2017-07-03:
Just a followup.  I actually changed the circuitry to use two different pins for buttons and the same thing happened.  Used PCINT 0 & 1, with 0 only working when made LOW and 1 only working when made HIGH.

Bwanna, 2017-07-03:
UPDATE:  it appears there's an issue with the pin NOT staying set when INPUT_PULLUP is used.  based on a suggestion, I added even more resistance to the ext pullup and it now works.  Both inputs activate the PCINT when put to LOW.  

Definately not a code issue for Nico, but does have something to do with the ATTiny85 parameters

NicoHood, 2017-07-03:
So the internal attiny pullups are too weak in this case while other atmegas are better at this point?

Bwanna, 2017-07-04:
That could be the issue.  A poster on the DigiSpark site said he remembered there was some type of coding fix and was going to see if he could find it.

NicoHood, 2017-09-02:
@Bwanna any updates on the issue? I will close it, as its no software bug from my site. Would be nice to hear the results.

closed 2017-09-02 by NicoHood