NicoHood/PinChangeInterrupt #19: Question on combining this with sleep function

mrohner, 2018-07-12:
Is the pinchangeinterrupt trigger a wake up from sleep? Do you issue a sei() somewhere or do I have to do it in the main program?
```
/*
ATTINY13A Pins
--------------
PB0: reed switch      over 10kOhm to GND
PB1: same reed switch resp. connected to PB0
PB2: 433Mhz sender
PB4: tempering switch over 10kOhm to GND

*/
#include <avr/io.h>
#include <avr/interrupt.h>
#include <util/delay.h>
#include <avr/sleep.h>
#include "PinChangeInterrupt.h"

volatile int message = 0;

const int SPEED = 350;
const byte START = 31;
const byte _SHORT = 1;
const byte _LONG = 3;
const byte OPEN =  0b00000001;//Decimal: 131586 (24Bit) Binary: 000000100000001000000010
const byte CLOSE = 0b00000010;//Decimal: 65793 (24Bit) Binary: 000000010000000100000001
const byte TEMPER = 0b00000011;//Decimal: 197379 (24Bit) Binary: 000000110000001100000011

#define TRANS_PIN PB2
#define INPUT_PIN0 PB0
#define INPUT_PIN1 PB1
#define TEMPER_PIN PB4

void setup(){
  // Configure TRANS_PIN as output
  DDRB |= (1 << TRANS_PIN);
  // Set INPUT_PIN as Input
  DDRB &= ~(1 << INPUT_PIN0 | 1 << INPUT_PIN1 | 1 << TEMPER_PIN);
  //pinMode(INPUT_PIN0, INPUT);//INPUT_PULLUP
  //pinMode(INPUT_PIN1, INPUT);
  MCUCR = (1 << SE)     // To enter any sleep modes, the SE bit in MCUCR must be written to logic one
        | (1 << SM1);   // power down sleep mode
  // attach the new PinChangeInterrupts and enable event functions below
  attachPinChangeInterrupt(digitalPinToPinChangeInterrupt(INPUT_PIN0), tick, RISING);
  attachPinChangeInterrupt(digitalPinToPinChangeInterrupt(INPUT_PIN1), tock, FALLING);
  attachPinChangeInterrupt(digitalPinToPinChangeInterrupt(TEMPER_PIN), temper, RISING);
  enable_PinChange_interrupts ();
}

void loop() {
  // change into sleep mode
  // Set TRANS_PIN low
  PORTB &= ~(1 << TRANS_PIN);
  ADCSRA = 0 << ADEN;   // Power reduction ADC
  sleep_cpu(); 
  ADCSRA = 1 << ADEN;   // Power reduction ADC reverse
  disable_PinChange_interrupts ();
  switch (message) {
    case 1:
      send(OPEN); 
    break;
    case 2:
      send(CLOSE);
    break;
    case 3:
      send(TEMPER);
    break;
  }
  message = 0;
  _delay_ms(50);
  enable_PinChange_interrupts ();  
}

void tick(void) {
  message = 1;
}

void tock(void) {
  message = 2;
}

void temper(void) {
  message = 3;
}

void enable_PinChange_interrupts () {
  enablePinChangeInterrupt(digitalPinToPinChangeInterrupt(INPUT_PIN0));
  enablePinChangeInterrupt(digitalPinToPinChangeInterrupt(INPUT_PIN1));
  enablePinChangeInterrupt(digitalPinToPinChangeInterrupt(TEMPER_PIN));
}

void disable_PinChange_interrupts () {
  disablePinChangeInterrupt(digitalPinToPinChangeInterrupt(INPUT_PIN0));
  disablePinChangeInterrupt(digitalPinToPinChangeInterrupt(INPUT_PIN1));
  disablePinChangeInterrupt(digitalPinToPinChangeInterrupt(TEMPER_PIN));
}

void send(byte id){
  for(byte i=0;i<13;i++) {
    sendStart();
    sendUniqueId(id);
  }
}

void sendStart() {
  digitalWrite(1);
  _delay_us((SPEED*_SHORT));
  digitalWrite(0);
  _delay_us((SPEED*START));
}

void sendUniqueId(byte id) {
  for(byte x=0;x<3;x++) {//  send 3 times same 8 bits = 24 bit code
    for (unsigned int y = 8; y-- > 0;){ 
      if(bitRead(id,y)) sendPulse1();
      else sendPulse0();
    }
  }
}

void sendPulse0() {
  digitalWrite(1);
  _delay_us((SPEED*_SHORT));
  digitalWrite(0);
  _delay_us((SPEED*_LONG));
}

void sendPulse1() {
  digitalWrite(1);
  _delay_us((SPEED*_LONG));
  digitalWrite(0);
  _delay_us((SPEED*_SHORT));
}

void digitalWrite(bool val){
  if(val)PORTB |= (1 << TRANS_PIN);
  else PORTB &= ~(1 << TRANS_PIN);
}
``

NicoHood, 2018-07-12:
Just try it. It should wake from my understanding. sei() is called by the Arduino IDE before setup() is called.

mrohner, 2018-07-13:
Yes, it works with a modification due to the ATTINY's prescaler. Thanks

See: https://github.com/mrohner/Door-sensor-with-433MHz-and-ATTINY13

closed 2018-07-13 by mrohner