NicoHood/PinChangeInterrupt #20: millis() issue

geologic, 2018-08-03:
I have 2 pushbuttons and want to record elapsed time between button1 press and button2 press. I used your library, inside each event function i use millis() to record time and in main code i print de elapsed time. Everything is working ok.

To preserve power, i added rocketscream/Low-Power library and put this line in the main loop:

LowPower.powerDown(SLEEP_FOREVER, ADC_OFF, BOD_OFF); 

Pushbutton events are still fired but elapsed time is not working, it always report less than 70ms, no matter time between button presses.

Than i tested without your library, using external interrupts (pins 2,3). millis() with lowpower library works as expected.

So, it seems that the problem with millis() happens when joining your library with lowpower library.
Any idea why?

NicoHood, 2018-08-04:
When you go into sleep, the CPU is suspended. Which means, the millis counter cannot increase. It sounds weird that the normal pin interrupts work properly, which too me sounds like it is a problem with them.

geologic, 2018-08-07:
Hum... i see.
I changed my code so when detects 1st button, does not go to sleep until detects 2nd button. This way i can use your library, sleep library and millis().

Thanks
Case closed.

closed 2018-08-07 by geologic