NicoHood/PinChangeInterrupt #22: PB6 and PB7

geologic, 2018-10-22:
Hi

I'm using a 328P on a breadboard and i set PB6 and PB7 as a digital I/O pins. I tested with a led and they are working ok.
Now i want to use those pins with PinChangeInterrupt, but they don't respond to changes. I read that i need to enable those pins on PinChangeInterruptSettings.h, but i don't know how to do it.
Can you please help?

Thanks

NicoHood, 2018-10-22:
Remove this:
https://github.com/NicoHood/PinChangeInterrupt/blob/master/src/PinChangeInterruptSettings.h#L123-L128

geologic, 2018-10-22:
I comented those lines, but no luck... 
They still don't respond to changes.

NicoHood, 2018-10-22:
Did you try PB5?

NicoHood, 2018-10-23:
Hm I dont know why.

geologic, 2018-10-23:
I found the problem. This is my original code: 

```
#define XTAL2 20     //XTAL is D20
attachPCINT(digitalPinToPCINT(XTAL2), Detected, CHANGE);
```

This doesn't work, but the following does:

`attachPCINT(6, Detected, CHANGE);`

So the problem is in your digitalPinToPCINT() doesn't map PB6 to IDE pin 20

NicoHood, 2018-10-24:
Okay, thanks for your help here. I am not sure if pin 20 is officially mapped to the XTAL2, is it? Maybe we could fix the macro?

geologic, 2018-10-24:
Well, i followed this to add support for XTAL1/XTAL2 as D20 / D21

[](http://web.archive.org/web/20130524110602/http://placeboaudio.com/how-tu-use-atmega-pb6-pb7-with-arduino-when-n)

So, i think is not official.

NicoHood, 2018-10-25:
Well then it is expected. I suggest using the direct PCINT number if you are using such an uncommon scenario. thanks for the feedback :)

closed 2018-10-25 by NicoHood