NicoHood/PinChangeInterrupt #35: How to enable at least one PCINT port and pin

ArminJo, 2021-01-03:
Hi Nico
I am stuck, I use your IRLremote library and have the signal attached at A0 PC0. Therefore I included <PinChangeInterrupt.h> before 
<IRLremote.h>  https://github.com/ArminJo/QuadrupedControl/blob/504561d053f1939da62ddb7e9c2d038050b99f22/src/IRCommandDispatcher.cpp.h#L72
The difference in code size between using pin 2 and pin A0 is now **500 bytes**. How can I restrict your PinChangeInterrupt library to support only Pin PC0 aka A0.
Using
```
#define PCINT_ENABLE_PCINT8
#define PCINT_ENABLE_MANUAL
```
leads to `#error Please enable at least one PCINT port and pin!`
But I do not know how to enable at least one PCINT port and pin  😟 
Why 
using
```
#define PCINT_ENABLE_PCINT8
#define PCINT_ENABLE_PORT1
#define PCINT_ENABLE_MANUAL
```
gives no error (I feel the define PCINT_ENABLE_PORT1 is redundant) but also does not save one byte.

BTW: Whats about [enabling discussions](https://docs.github.com/en/free-pro-team@latest/github/administering-a-repository/enabling-or-disabling-github-discussions-for-a-repository#enabling-or-disabling-discussions-for-your-repository)?

NicoHood, 2021-01-03:
I've enabled the discussions (good suggestion), but how can i transfer this issue?

What board are you using? The library itself works properly, but now you want to make the code size even smaller, correct?

ArminJo, 2021-01-03:
> I've enabled the discussions (good suggestion), but how can i transfer this issue?
I guess transfering is not possible.

I have a Nano board. I used your NEC IR library together with the PinChangeInterrupt library, but wanted to switch to my own NEC library and keep the PinChangeInterrupt. But now I build my own Interupt handler and saved around 750 bytes compared to the former solution.

closed 2021-01-03 by ArminJo