NicoHood/PinChangeInterrupt #39: Ajout ATTinyx61

astro59, 2021-05-16:
Est-il possible d'utiliser PinChangeInterrupt avec des ATTiny861 ?

NicoHood, 2021-05-16:
Not yet, but I can add support for it, if you like. It seems to support PCINT on all pins:
http://ww1.microchip.com/downloads/en/devicedoc/atmel-2588-8-bit-avr-microcontrollers-tinyavr-attiny261-attiny461-attiny861_datasheet.pdf

astro59, 2021-05-18:
Yes, I like it. I will be happy to test this on my project. I need a lot of independent pins, a few of which are interrupts.

NicoHood, 2021-05-18:
I accidentally commited the changes directly to master, without a PR. The commit is this one:
https://github.com/NicoHood/PinChangeInterrupt/commit/ca4fe2e228b71adc87bec0402c12db720f58396e

Please test and report if it works. Please test all pins (not just a few), if possible.

NicoHood, 2021-05-18:
I've just released 1.2.9. Please test if that works for you. I expect that this should be an easy one and just work. If not, I will try to fix it ASAP.

astro59, 2021-05-18:
Hi,

I tested the library but without success.

I just replaced LED_BUILTIN by pinLED to place a led with its resistor on an
ATTiny861.

I attach a photo of the assembly and the modified source code.

 

The led lights up for 1 second but the button does not work

 

Did I make a mistake?

 

Thanks

 

 

  _____  

 

<https://github.com/NicoHood/PinChangeInterrupt/issues/39#issuecomment-84338
KQFRANCNFSM4463HNPA> .
<https://github.com/notifications/beacon/AI7YQE4LPRSOGUCZNJCQ3JDTOKQFRA5CNFS
M4463HNPKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOGJCQQEA
.gif>

NicoHood, 2021-05-19:
I cannot see a photo. Please paste your source, if possible.

This is the example to use for testing:
https://github.com/NicoHood/PinChangeInterrupt/blob/master/examples/PinChangeInterrupt_Led/PinChangeInterrupt_Led.ino

astro59, 2021-05-19:
```
#include <PinChangeInterrupt.h>
#include <PinChangeInterruptBoards.h>
#include <PinChangeInterruptPins.h>
#include <PinChangeInterruptSettings.h>

// ATMEL ATTINY 261/461/861 et 261V/461V/861V  -  BRANCHEMENTS

// Choose a valid PinChangeInterrupt pin of your Arduino board
#define pinBlink 4
#define pinLED   11

void setup() {
  // set pin to input with a pullup, led to output
  pinMode(pinBlink, INPUT_PULLUP);
  pinMode(pinLED, OUTPUT);

  // Manually blink once to test if LED is functional
  blinkLed();
  delay(1000);
  blinkLed();

  // Attach the new PinChangeInterrupt and enable event function below
  attachPCINT(digitalPinToPCINT(pinBlink), blinkLed, CHANGE);
}

void blinkLed() {
  // Switch Led state
  digitalWrite(pinLED, !digitalRead(pinLED));
}

void loop() {
  // Nothing to do here
}
```
`

astro59, 2021-05-19:
Oups !

for this the button is on physical pin 14 and led on physical pin 4
I have tested other configurations.

astro59, 2021-05-19:
`[https://www.astrolynx.com/Electronique/IMG_20210518_202605.jpg`](url)

NicoHood, 2021-05-19:
can you check what `digitalPinToPCINT(pinBlink)` returns?

astro59, 2021-05-19:
I have to go
I'm doing this tonight when I get home

astro59, 2021-05-19:
IDE 1.8.13
ATTinyCore Spence Konde 1.5.2
Apparently C:\ ... \Arduino15\packages\ATTinyCore\hardware\avr\1.5.2\variants\tinyX61_New\pins_arduino.h

Tested with 4, PA4 and PCINT4
Start up the led lights up for 1 second
Pin 14 (PCINT4)   :    Button released 5V   -   Button pressed 0V   - No réaction

in digital mode, all is OK

NicoHood, 2021-05-20:
What does digitalPinToPCINT(pinBlink) return?

astro59, 2021-05-20:
I don't know.
how i do that?

NicoHood, 2021-05-20:
Something like:
```
if (digitalPinToPCINT(4) == 4){
led on
}
```

if it is not true, please check against -1 or try & error all other numbers.

tredpath, 2021-06-25:
```
digitalPinToPCINT(0) == 1
digitalPinToPCINT(1) == 2
digitalPinToPCINT(2) == 4
digitalPinToPCINT(3) == 8
digitalPinToPCINT(4) == 16
digitalPinToPCINT(5) == 32
digitalPinToPCINT(6) == 64
digitalPinToPCINT(7) == 128
digitalPinToPCINT(8) == 9
digitalPinToPCINT(9) == 10
digitalPinToPCINT(10) == 12
digitalPinToPCINT(11) == 16
digitalPinToPCINT(12) == 24
digitalPinToPCINT(13) == 40
digitalPinToPCINT(14) == 72
digitalPinToPCINT(15) == 136
```

There aren't separate interrupt vectors for each port either. ISR(PCINT_vect) covers everything (see table 9-1 in the [datasheet](https://ww1.microchip.com/downloads/en/DeviceDoc/Atmel-2588-8-bit-AVR-Microcontrollers-tinyAVR-ATtiny261-ATtiny461-ATtiny861_Datasheet.pdf) for the list of interrupt vectors).

NicoHood, 2021-07-09:
Great progress on your PR so far! I want to note, that we need to also identify the root cause of the invalid `digitalPinToPCINT()` mappings.

NicoHood, 2021-07-09:
Which Core are you using to support this attiny? I need to check if they defined every marco correct.
https://github.com/NicoHood/PinChangeInterrupt/blob/d604e1f0bddb0f67fab8d3da753b48d80be02db1/src/PinChangeInterrupt.h#L103