NicoHood/PinChangeInterrupt #40: Please consider adding support for member function ISR registration

SteveGuidi, 2021-05-19:
Currently, the `attachPCINT()` macro accepts a pointer to an argument-less void function, which makes it cumbersome to pass a non-static member-function pointer.  We could use `std::bind` and other library features to get around this, but in general, the C++ Standard Library is inaccessible to most Arduino architectures.

It would be great if `attachPCINT()` (or other construct) had an overload that accepted a single `void*` parameter in addition to the callback function pointer as arguments, then called the ISR with the given `void*` parameter as the sole argument.  This should allow for binding to non-static member functions.

I'm sure I can eventually produce a pull request for such a feature, but I'm currently wrapping my head around the casts and function pointer syntax required to do so and am getting very confused.

NicoHood, 2021-05-19:
Cant you just attach a "normal" function, like you currently do and inside this function do the described call? I do not understand the benefit, but I have not been coding c++ for a while and I remember there were some usecases. So please go into detail, I am happy to integrate that, if you come up with a reasonable solution :-)

SteveGuidi, 2021-05-19:
Yes, that will certainly work.  In my case though, I'd like to use a member function as the callback as it manipulates class instance data and keeps the code encapsulated.

For reference, the member function is non-static, so the compiler adds an implicit first argument which is the pointer to the object making the function call.

Here is an example showing the semantics:

```
void globalISR() { /* do work */ };

class C {
    public:
        static void DoWork() { /* ... */ };
        void DoOtherWork() { _data = 123; };

        void AttachISR() {
            attachPCINT(digitalPinToPCINT(4), globalISR, RISING);  // Works
            attachPCINT(digitalPinToPCINT(5), DoWork, RISING);  // Works
            attachPCINT(digitalPinToPCINT(6), DoOtherWork, RISING);  // error: signature is void(*)(C*), not void(*)()
        };

    private:
        int _data;
};
```

The trick to supporting the member function registration is to produce a wrapper function that is compatible with the macro, but store the `this` argument somewhere so that the wrapper can pass it to the proper ISR.  This is basically what `std::bind` will do for us, something like the following.

```
// Function pointer and operator syntax not checked.
struct MemberISRWrapper {
    void* isrObject;
    void (*callback)(void*);  // void callback(void* arg);

    void operator() () {
        callback(isrObject);
    }
};

// Using class C above
class C {
    private:
        MemberISRWrapper _wrapper { this, &DoOtherWork };  // pseudo-code shortcut-initialization -- syntax errors likely.

        void AttachISR() {
            // ...
            attachPCINT(digitalPinToPCINT(6), _wrapper, RISING);  // _wrapper implicitly converts to (void*)()
        };
};
```

Sorry, this is a lot of text!  But hopefully it gives some context as to what I'm trying to accomplish.  The struct/wrapper solution above that emulates `std::bind` is probably not suitable for the library, but I'm hoping that there is a macro-level equivalent that might work.  Otherwise, clients will need to implement a similar bind-like structure to accomplish this.  My understanding of macro-programming is very poor, but I may make some headway after further study!

NicoHood, 2021-05-19:
Oh, so you want to attach the pcint within a class?

This is how I am doing it in the IRL library:
https://github.com/NicoHood/IRLremote/blob/master/src/IRL_Receive.h#L110
https://github.com/NicoHood/IRLremote/blob/master/src/IRL_Receive.h#L61-L64

I am not sure if that helps, as its been very looong since I've written that. I dont remember all details neither.

SteveGuidi, 2021-05-21:
Yup, attaching a PCINT within a class -- but more precisely, to a non-static member of the class.  The example you linked attaches to a static member function.

https://github.com/NicoHood/IRLremote/blob/master/src/IRL_Receive.h#L63

I've been hacking away at this to try get my pseudo-code from [above](https://github.com/NicoHood/PinChangeInterrupt/issues/40#issuecomment-844390581) to work, and I keep encountering different obstacles.  I took a step back and realized that the native Arduino external interrupts follow the same global or static function semantics, and that almost all forum messages for using interrupts with member functions suggest using global or static functions to avoid the issues I am experiencing.

I think some other infrastructure is needed before any library attempts to support this, if it is even doable.  So with that said, I can live with static functions for now and I may revisit this in the future to take another crack at.

Thank you for your patience and consideration!

closed 2021-05-21 by SteveGuidi