NicoHood/PinChangeInterrupt #41: Attiny13a, MicroCore and PinChangeInterrupt problem.

cbpdk, 2021-06-20:
Attiny13a, 9.6 Mhz
MicroCore 2.0.3
PinChangeInterrupt 1.2.9
Arduino IDE 1.8.15

Trying MicroCore 2.0.3 and the PinChangeInterrupt 1.2.9 library in connection with the Attiny13a, I ran into problems.

Of course, I might have overlooked something obvious as this it the first time I tried pin change interrupt
on an attiny13a.

No interrupts were caught.
After modifying and testing PinChangeInterrupt_HowItWorks.ino on the attiny13a, at first nothing happened.
In the attiny13a documentaion, page 47, it says:

When the PCIE bit is set (one) and the I-bit in the Status Register (SREG) is set (one), pin
change interrupt is enabled.

Skimming though the PinChangeInterrupt sources, I did not come across enabling of the pin change interrupt.

And no option to enable pin change interrupt in MicroCore core_settings.h

Adding a sei(); after the attachPCINT function(s), it finally worked.

`

#include "PinChangeInterrupt.h"

// Choose a valid PinChangeInterrupt pin of your Arduino board
#define pinBlink 2
#undef LED_BUILTIN
#define LED_BUILTIN 4

void setup() {
  // set pin to input with a pullup, led to output
  pinMode(pinBlink, INPUT_PULLUP);
  pinMode(LED_BUILTIN, OUTPUT);

  // Manually blink once to test if LED is functional
  blinkLed();
  delay(1000);
  blinkLed();

  // Attach the new PinChangeInterrupt and enable event function below
  attachPCINT(digitalPinToPCINT(pinBlink), blinkLed, CHANGE);

  **interrupts();    //Or sei(); needed here to enable interrupts.**
}

void blinkLed(void) {
  // Switch Led state
  digitalWrite(LED_BUILTIN, !digitalRead(LED_BUILTIN));
}

void loop() {
  // Nothing to do here
}

`

Testing using three attachPCINT and serial output, it still left memory enough to catch data and transmit.
Well done.

NicoHood, 2021-06-20:
Looks like an issue of the core itself?

cbpdk, 2021-06-21:
Yes, this was a core issue. The updated core and PinChangeInterrupt now work as expected.

closed 2021-06-21 by cbpdk