NicoHood/PinChangeInterrupt #46: Add Support For Arduino Nano RP2040 Connect

ZachEnglish, 2021-07-13:
The 2040 has a lot more interrupt functionality built-in, which is handy, but I have a fairly large project already in place that uses this library on a "regular" Nano and would like to swap out to use a Nano RP2040 without a lot of code changes. It would be quite convenient if PinChangeInterrupt would continue to work when building for that target.

Thanks for the library, by the way! This has been a life saver for my own projects :-)

NicoHood, 2021-07-13:
That board uses an ARM, but does ARM has PCINTs at all? I am sure normal interrupt will be sufficient here and this library is built only for AVR.

ZachEnglish, 2021-07-13:
Nope, you are 100% correct, for someone starting a new project with just the RP2040 as the mainboard they have no reason to use this library at all. My catch is that I have an already existing project that uses an AVR board and this RP2040 board is a drop in replacement (basically) which adds WiFi, Bluetooth, a gyroscope, a microphone, and WAY more RAM (2k usable to 264k usable). Updating my project to support both boards will require some extra abstraction layers when it comes to interrupt handling, and if at all possible, I'd prefer to have that baked into the library... but if that isn't the calling/purpose/intent of this library, that's totally fine :-) I may poke around and see if the problem is easier for me to solve as a pull request against this library or in my own code. Let me know if you'd be open to that as a solution. Thanks again!

P.s. On a completely different note, but something that may be of interest to you, when listing PinChangeInterrupt as a dependency in another library for some reason it isn't found. I've tried a few times a few different ways (and all the other libraries I use work just fine). Not sure if that is a library configuration issue (I didn't notice anything off when I looked) or an Arduino Library Manager issue. I noticed because my library (which lists PinChangeInterrupt as a dependency) was used in an intro to CS high school classroom environment and anything that didn't just happen super smoothly for the students caused some headache. Everyone had to install PinChangeInterrupt on its own instead of it getting pulled through with the install of my library. You can see my library.properties file here: https://github.com/ZachEnglish/NanoProtoShield/blob/master/library.properties . Anyway, that's more of just a heads up! Thanks again for making such a useful tool! I can't believe it is such a secret, very few tutorials mention you can get such awesome interrupt behavior out of an AVR board with such ease as your library provides.

NicoHood, 2021-07-13:
About the first comment: I'd be happy to look at any solution you provide. Just open a PR or comment here again. For better issue tracking I will close this issue though. You can still comment.

About the dependency issue: It would be nice if you can post that in the arduino forum or bug tracker and find out the root cause. Maybe @per1234 can help? That should of course not happen, but I am not the right person to ask. Thanks for reporting (I did not even know my library was used for educational purposes, nice)!

closed 2021-07-13 by NicoHood

per1234, 2021-07-13:
Hi @NicoHood and @ZachEnglish. I investigated the issue with the NanoProtoShield library's dependencies installation and submitted a PR with a fix:
https://github.com/ZachEnglish/NanoProtoShield/pull/2

The problem was with a dependency in the list before the PinChangeInterrupt library dependency, which meant the installation process errored out before it got to the installation of this dependency.

ZachEnglish, 2021-07-13:
Thank you @per1234 ! I'll pull that change in, test it out, and then keep an eye on the LiquidCrystal I2C project issue so I can change it back if they fix their name....

Sorry for bugging you @NicoHood with what was apparently my own issue, but thank you very much for point me in the right direction :-)

NicoHood, 2021-07-14:
Nice to hear! thanks a lot!