NicoHood/PinChangeInterrupt #59: Conflict with SoftwareSerial.h

tcklab, 2023-05-14:
Compiling .pio\build\uno\src\PinChangeInterrupt_Led.ino.cpp.o
Linking .pio\build\uno\firmware.elf
SoftwareSerial.cpp.o (symbol from plugin): In function `SoftwareSerial::read()':
(.text+0x0): multiple definition of `__vector_3'
.pio\build\uno\src\PinChangeInterrupt0.cpp.o (symbol from plugin):(.text+0x0): first defined here
SoftwareSerial.cpp.o (symbol from plugin): In function `SoftwareSerial::read()':
(.text+0x0): multiple definition of `__vector_4'
.pio\build\uno\src\PinChangeInterrupt1.cpp.o (symbol from plugin):(.text+0x0): first defined here
SoftwareSerial.cpp.o (symbol from plugin): In function `SoftwareSerial::read()':
(.text+0x0): multiple definition of `__vector_5'
.pio\build\uno\src\PinChangeInterrupt2.cpp.o (symbol from plugin):(.text+0x0): first defined here
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\uno\firmware.elf] Error 1

zaindroid, 2024-02-26:
Did you find solution for this?

Jediweirdo, 2024-11-19:
To update you guys on this, apparently SoftwareSerial uses delays in its interrrupts (which is a pretty bad coding practice), which makes it incompatible with most Interrupt-Based Libraries, like this one. You can read more about it [here](https://github.com/GreyGnome/EnableInterrupt/wiki/FAQ#no-really-the-enableinterrupt-library-does-not-work-with-the-software-serial-library), but the TL;DR of all this is to stop using SoftwareSerial and try using its counterpart [AltSoftSerial](https://github.com/PaulStoffregen/AltSoftSerial) if you aren't using all of your Arduino's timers. If you are... I guess you're kinda cooked :p